chris1111/Wireless-USB-Big-Sur-Adapter #323: Any way to support RTL8192FU?

eeneecha, 2024-07-15:
I found a driver from the manufacturer's website but the driver only supports up to Mojave.  [https://www.tendacn.com/download/detail-2911.html](url)

chris1111, 2024-07-15:
Did you try [ Wireless-USB-Big-Sur-Adapter
 ](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter)

eeneecha, 2024-08-04:
> Did you try [ Wireless-USB-Big-Sur-Adapter](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter)

Yes. I have tried it. It stays on "No usb wifi-plugin"

chris1111, 2024-08-04:
> > Did you try [ Wireless-USB-Big-Sur-Adapter](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter)
> 
> Yes. I have tried it. It stays on "No usb wifi-plugin"

There is [no source for the Driver](https://github.com/chris1111/Wireless-USB-Big-Sur-Adapter/discussions/209)

closed 2024-08-04 by chris1111