chris1111/Wireless-USB-Big-Sur-Adapter #334: WiFi USB is recognized, but the app doesn't show any available network

BenjaminStonawski, 2024-09-26:
Hello there!
I installed this utility on my Early 2015 MacBook Pro running macOS Ventura. I'm using a TP-Link WN823Nv2 WiFi USB adapter. After installing the app, the utility recognizes the adapter but doesn't show any available network (shown in the screenshots below). What could be the issue? 
Thank you for your answer in advance!
[image]
[image]

chris1111, 2024-09-26:
Try turn Off and On?
Also check your router

BenjaminStonawski, 2024-09-26:
Yes, I tried to restart the Mac several times and tried different USB ports also. There are more WiFi routers in the area (2-3) but none of them is showing up.

chris1111, 2024-09-26:
First time seeing this you are certainly using OCLP?

BenjaminStonawski, 2024-09-26:
Yes, I'm using OpenCore Patcher. Maybe should I try to reinstall root patches?

chris1111, 2024-09-26:
Uninstall this utility 
Reboot Mac OS then try the one dedicated for OpenCore users

BenjaminStonawski, 2024-09-27:
I tried that but same error occurred.
I even tried to reinstall root patches and update OpenCore but didn't fix the issue either.

chris1111, 2024-09-27:
Check your config.plist, some thing wrong probably enter in conflicts with the utility because I have no issue same adapter from 10.15 to macOS 15

BenjaminStonawski, 2024-09-27:
What could conflict? Are there any specific boot-args that need to be added?

BenjaminStonawski, 2024-09-27:
Does SIP need to be fully disabled? Because after entering "csrutil disable" Terminal app says "SIP status: unknown".

closed 2025-02-20 by chris1111