chris1111/Wireless-USB-Big-Sur-Adapter #337: NO USB-WIFI PLUG IN - TPLINK WN823N V2.0

xingzj, 2024-10-19:
my wn823n v2.0 can't be found on my intel hackintosh, no usb-wifi plug in was appeared, i think the driver and config.plist is correct because my another rtl 8822bu is drivered and everything is ok, is there any special step for wn823n v2.0? thanks

xingzj, 2024-10-19:
[image]

chris1111, 2024-10-19:
it's strange because I have the same wifi adapter but mine has a network search button on the other hand mine works very well and it's also the V2

[image]

xingzj, 2024-10-21:
I see，so it should be vary version in defferent region, mine is mainland China，and it's no driver version, it would appeared as a optical drvier firstly in Win system and when the driver program was installed the optical driver will disappeared, and then the USB
Wifi Stick will be working

chris1111, 2024-10-21:
I will close this issue at this time

closed 2024-10-21 by chris1111